MickGyver/DaemonBite-Retro-Controllers-USB #10: NES adapter recognized in Win10 but not responding

quaelgeist, 2020-09-08:
Hi guys.

I have wired up the NES adapter and it shows up as two gamepads in win10.
But a confirmed as working NES pad produces no response from any of those two.

Has anybody ever had this problem before?

regards,
Michael

MickGyver, 2020-09-08:
Probably something wrong with the wiring. Did you connect all wires of the connector, including 5V and GND?

quaelgeist, 2020-09-08:
thanks for the answer

Yes, all are connected. latch, clock, gnd, data1 and vcc (thats the 5V, right?)
also created the solder bridge and checked the wires of the extension plug with a multimeter to make sure i wasn't mixing them up. Solder points look and feel solid.

quaelgeist, 2020-09-08:
Well, reproduce me sideways, but after trying yet again, it suddenly works perfectly. I guess it was Windows and not the Arduino that was pulling my leg here all the time :-|

closed 2020-09-08 by quaelgeist

MickGyver, 2020-09-08:
Good to hear it's working! :)