MickGyver/DaemonBite-Retro-Controllers-USB #12: Could you make one for the Nintendo classic connector

aihooo444, 2021-03-08:
Not a Issue but a feature request.
I would love to use my Official Nes Mini Controllers

asturur, 2021-03-29:
I have the same whish and i noticed that the controllers seems to be based on an i2c bus at 400khz.
Does this help to understand if is doable or not?

Klausus, 2021-04-28:
> 
> 
> I have the same whish and i noticed that the controllers seems to be based on an i2c bus at 400khz.
> Does this help to understand if is doable or not?

All the classic controller I2C timing is measured with SNES Classic Mini (I2C speed is 200KHz).

Some non-official classic controller (like PS2 to classic controller) need 200KHz speed, official one can run with 300KHz.

But both can not work properly with 400KHz speed.

**Note:**  This info is extracted from this Github Project: https://github.com/splash5/arduino_classic_controller_to_snes

Klausus, 2021-04-28:
**It is possible, but I don't know the lag it may have.**
Exists implementations made in arduino and there are also entire libraries dedicated to only connect Wii/ClassicMini I2C controllers.

**For example:**

Wii Classic controller to SNES interface:
https://github.com/splash5/arduino_classic_controller_to_snes

Library 1 (older):
https://github.com/madhephaestus/WiiChuck

Library 2 (more newer):
https://github.com/dmadison/NintendoExtensionCtrl

Example of usb interface with arduino for 1 classic controller:
https://www.instructables.com/USB-Wii-Gamepad-Using-Arduino-Leonardo/

The libraries have support for many extra peripherals that are not needed (nunchuck, guitar hero, taiko drums etc) and I don't know how much it could affect to minimize input lag.

We need to make an implementation that allows to connect 2 classic controllers of Wii or NES/SNES Classic Mini, minimizing the lag as much as possible.

@MickGyver  Could you check it out when you have time?

PD:  Sorry if English is not perfect (My native language is Spanish). Translated with Deepl.

Thanks.

mbaran5, 2021-06-06:
> We need to make an implementation that allows to connect 2 classic controllers of Wii and NES/SNES Classic Mini

My understanding of the way these are setup, the i2c address for all classic controllers is the same, and is not able to be changed. This would require external hardware to the pro micro (a i2c multiplexer) otherwise they'd appear as the same device on the i2c bus if both connected.

That said, I would love to see support for the Classic controllers as the quality of the real Nintendo ones is high and they'd be great to use with this.

asturur, 2021-06-07:
one controller per arduino would be cool enough. as of today an extension cord to cut to create the female port is more expensive than the arduino 🤷‍♂️