MickGyver/DaemonBite-Retro-Controllers-USB #18: Problem with a 6 buttons megadrive controller

blc29seb, 2021-04-28:
Hi 
I made the controller adapter for megadrive (SegaTwoControllersUSB)
tested under windows 10 with the windows game controller application 
i have several problems

I use an official 6 buttons controller and an official 3 buttons
in 3 buttons mode everything works fine (even with the 6 buttons controller configured in 3 buttons (mode button))
but in 6 buttons mode when I use the Dpad I have some buttons that are activated (4,5,6,8,9)
for example (left +down):
[image]

and when I push on the x (4) y(5) or z(6) buttons they blink instead of staying on  like A,B and C buttons

What does the button 9 correspond to when there are only 8 buttons ?
Sincerely (sorry for my bad english ;) )

MickGyver, 2021-05-06:
I guess there is something with your wiring/soldering is causing this issue. 

Button 9 corresponds to the HOME button on the 8bitdo M30 2.4G wireless MD/Genesis controller.

venice1200, 2021-05-06:
Using your "SegaControllerUSB" Sketch.

My two Sega 3 Button Controllers Model 1650 and 1650-50 are working fine.
My Sega 6 Button Controller Model SJ-6000 doesn't work at all.

Any Idea's?

blc29seb, 2021-05-09:
Hi 
It's not a problem of welding or anything else 
I checked everything
and with this code (https://github.com/thinghacker/DualUSBSegaController/tree/master/examples/DualUSBSegaController)
 it works fine but I can't use your TX1 and RX0 connections 
I will have to redo my soldering to wire them on other ports 
Best regards

venice1200, 2021-05-12:
Just a note, I got my 8BitDo Genesis M30 2.4g Controller whioch works perfect with the "SegaControllerUSB" Sketch. Really cool.

Need to double check why the Sega 6 Button Controller Model SJ-6000 doesn't work at all.
I will try it with the MiSTer's USB 3.0 based Snac Adapter next week.

MickGyver, 2021-05-13:
> Using your "SegaControllerUSB" Sketch.
> 
> My two Sega 3 Button Controllers Model 1650 and 1650-50 are working fine.
> My Sega 6 Button Controller Model SJ-6000 doesn't work at all.
> 
> Any Idea's?
Please try the "SegaControllerUSB" sketch in this branch:

https://github.com/MickGyver/DaemonBite-Retro-Controllers-USB/tree/SEGA-6-Button

venice1200, 2021-05-13:
Tested. Nothing better with Sega 6 Button. No function at all.
And from the 8BitDo Genesis M30 2.4g only Buttons ABC and Start are working.

MickGyver, 2021-05-13:
> Tested. Nothing better with Sega 6 Button. No function at all.
> And from the 8BitDo Genesis M30 2.4g only Buttons ABC and Start are working.
> Back on "Main".

Yes, I just noticed it was a stupid attempt at a fix, I will look at it some more when I have the time.

venice1200, 2021-05-14:
No hurry...

gregdick1, 2021-07-07:
I've run into a similar problem with my SegaTwoControllersUSB adapter and the retro-bit official 6 button controllers. Without pressing anything, x or y will fire seemingly randomly.
I made two adapters. They both behave the same. They work perfectly with an original 3 button controller and with some non-branded cheap 6 button controllers I got off of ebay. Very interesting that the cheap one works fine. I don't have an original 6 button to try.

I also tried the SegaControllerUSB with 6 button hack on your other branch. Then both non-branded and retro-bit 6 button controllers only had the dpad, A, B, C, and Start working.

Jmaxime89, 2021-09-24:
Hello I'm wondering if there is any development on this ? I only have original 6 button controllers and they don't work on the two daemon bite I've made for the mister

quidamphx, 2021-09-29:
I have similar issues, which led me to double check my wiring.

I'm using MiSTer input test, and Windows 10 USB Game controller properties.

I wonder if this issue is somehow related to device IDing? When I have Player 1 connected with a 2.4Ghz 8bitdo controller, it works perfectly. When I add another receiver to Player 2, it causes sporadic inputs on Player one.

When I connect an OEM Genesis 1650 as Player 2 (still with the 8bitdo as Player 1), both function correctly. Narrowing it down, either it's getting confused by 2 controllers that are the same, or something related to the XYZ of Player 2.

If I knew more about the coding I'd look into it, but I can say there's an issue with Player 2 using 6 button, OR controllers being ID'd as the same.

quidamphx, 2021-09-30:
> Interesting, I will try my wired licensed retrobit 6 button with a OEM
> controller plugged in. It would be very interesting to understand why this
> particular controller doesn't work with the DaemonBite but in all of my
> Sega Genesis (Genesises?) they work flawlessly

Yeah, it's puzzling for sure. I don't have a Sega-branded 6 button to try,
but I can say for certain that there's issues with using my 2x M30
controllers together.

If there's any more testing I can do to help narrow it down, I'd be happy
to.

I'm repeating myself a bit, but the Player 2 controller works perfectly if
it's the only one plugged in (nothing in P1 spot), but with them both
8bitdo, I can't get anything to trigger in the P2 spot at all (and it
causes issues with P1 as mentioned), until I use a different controller.
Then it, and P1 work perfectly.

That said, the NES version I built works perfectly with 2x 8bitdo
controllers, and so does the SNES version.

Tom

emarleau, 2021-10-09:
I would like to point out that the issue with the Retro-Bit 6 button controller is also happening with the SNAC adapter.
[image]

emarleau, 2021-10-09:
Genesis core. I'm using a test rom called Sega Genesis 6 button controller test https://www.romhacking.net/homebrew/126/

treeswithhumanfaces, 2022-03-09:
I was able to fix this for retro-bit officially licensed controllers by commenting out the code that captures home button input.

Lines 162 and 234 in SegaControllers32U4.cpp

gregdick1, 2022-08-21:
@treeswithhumanfaces I tried commenting those two lines, but it didn't fix the problem for my retro-bit controllers. Did you have any other changes you experimented with that might've gotten it working?

diegov-au, 2026-03-22:
I was able to get this working with the Retro-Bit 6 button controller by reducing the scan cycle to 5 micro second in the SegaController32U4.h. e,g. SC_CYCLE_DELAY=5. I also recommend commenting out lines 162 and 234 as shared above for cleaner triggering with the Retro-Bit.