MickGyver/DaemonBite-Retro-Controllers-USB #31: Colour coded SNES connector diagram, USB C pro micro support

danboid, 2023-03-21:
Hi MickGyver

I think it would be useful to colour code the SNES pad diagram to use the colours used by most SNES pads and extension cables but I'm not sure if they differ from the colours used by NES pads?

I've done a quick hack job of making those adjustments to the labels on SNES pic that you probably won't want to use but here it is anyway:

[image]

Also, Amazon sell a USB C version of the Arduino Pro Micro. I presume we can use them instead of Micro USB version of the Pro Micro? I realise it won't be any faster, not that it matters, but I prefer the reversable nature of USB C.

Thanks for the great project!

danboid, 2023-03-25:
I have successfully built and tested this adapter using a USB C Pro Micro so you may want to update the README to mention that this is a option for those that might prefer to use USB C.

EmiFanshi, 2024-07-19:
Thank you very much. I had doubts until I found your comment. Thank you so much.🎀💗