MickGyver/DaemonBite-Retro-Controllers-USB #33: Hyperkin GN6 Sega Genesis 6 Button Controller Support

ZachHazen, 2023-12-11:
Testing the two controller USB Genesis Adapter

With a sega branded 3 button controller, all of the inputs work: (D-Pad, start, ABC)
With a Hyperkin GN6 6 button controller (works fine on console), only The D-Pad and the BC buttons work.  Start, A, mode, and XYZ do not work.

Tested on windows with gamepad properties and on a Mister.  Tested with two different GN6 controllers and a single OG genesis controller.

Checked pin outs with a multimeter and everything looks good.  Used arduino IDE to upload the latest sketch.  

Need to get an official sega 6 button controller to test more, but I wanted to check in here to see if there were any ideas.  The Hyperkin GN6 is a great alternative to aging controllers that have gone mushy, and they work fine on console.  Curious if there are any reasons why the adapter might see it differently.

ZachHazen, 2023-12-11:
I should add the GN6 works great on the Mister with SNAC.  Genesis SNAC/X is limited to single player, so I got a daemonbyte adapter specifically to play two player games.

ZachHazen, 2023-12-11:
I'm semi-consistently able to get the start button to register (button 7) by holding start (nothing happens), hitting B, releasing B (start comes on).  Its not every time...

[image]

ZachHazen, 2023-12-17:
Tested two OG Genesis 6 button controllers.  While I got them to work eventually, I did have a few moments where they were acting like the Hyperkins - no A, X,Y,Z or start.  Plugging them out and in again seemed to fix it?  ~1 hr of gunstar heros on the Mister + CRT felt perfect with 2 players.

darylrichards, 2023-12-22:
I've been working through the code, and have a version that works 95% on these Hyperkin units. I've tested them and a normal Atari joystick, and they both work fine. Only oddity so far is that the X,Y,Z buttons seems sticky. If I press one of them, then A,B or C, the XYZ sticks until the ABC is released... I'm away for the next week so I'll let someone else see what they can..

[SegaTwoControllersUSB.zip](https://github.com/MickGyver/DaemonBite-Retro-Controllers-USB/files/13747148/SegaTwoControllersUSB.zip)

Sorry, my Git-fu is lacking to get this up here easier..